Willmish/bibop-zephyr-eos-s3 #12: PRs needed for full EOS S3/Quickfeather support on Zephyr and others we can pick up

Willmish, 2023-06-08:
For Interrupts to work on Zephyr with EOS S3 (and so that SPI can work?): https://github.com/zephyrproject-rtos/zephyr/pull/28291

For OpenOCD flashing support to work on Zephyr: https://review.openocd.org/c/openocd/+/6956 https://github.com/Willmish/openocd/tree/eoss3_flash_patch

Incomplete support for quickfeather_Hal module for zephyr: https://github.com/zephyrproject-rtos/hal_quicklogic Antmicro's repo with seemingly some additional not moved over support (i2c, spi, etc): https://github.com/antmicro/eos-s3-hal

For YoSys (synthesis of FPGA images support for QuickLogic devices): https://github.com/YosysHQ/yosys/pull/1816

JDuchniewicz, 2023-06-08:
How is I2C handled? is it a hardware setup? code in FPGA? (probably hardware regisers etc).
How can we access Wishbone and I2C

Willmish, 2023-06-10:
NEXT THING:
In the quickfeather dts, Button0, blue green and red gpios are defined, but their pinmuxes are not set up in board.c (https://github.com/zephyrproject-rtos/zephyr/blob/533c1985dc18177d8986fea86f00d39c132cda5e/boards/arm/quick_feather/board.c#L15) only uart is.

See:
- https://github.com/zephyrproject-rtos/zephyr/blob/533c1985dc18177d8986fea86f00d39c132cda5e/soc/arm/quicklogic_eos_s3/soc_pinmap.h#L13
- https://github.com/zephyrproject-rtos/hal_quicklogic/blob/ba7e35f175fcd28199de11f955cb995e1ef20c73/HAL/inc/eoss3_dev.h#L1406
- And the technical reference manual page 170: https://www.quicklogic.com/wp-content/uploads/2020/06/QL-S3-Technical-Reference-Manual.pdf

Willmish, 2023-06-14:
- How can we utilise SensiML ? And how dependant is it on qorc-sdk?
- How to program the Sensor Manager? (We are currently reading data from I2C to M4, can we do some preprocessing in the Sensor Manager?
- Have you ever/do you have any tools for synthesizing the ML models to FPGA ? (then the data flow would be : senor over i2c -> FFE -> Sensor Manager -> FPGA -> M4)
- What else can we use the FPGA for ?

Willmish, 2023-06-14:
Turns out the pinmux was already set in bootloader, since removing gpio_pin_configure and the pinmux still meant the button was read fine, at least we learned:
- Our pinmux was correct  (otherwise, the button reading would fail now)
- There is something wrong in the dts, since gpio_pin_configure_dt causes the button GPIO to stop working properly

JDuchniewicz, 2023-06-14:
> NEXT THING: In the quickfeather dts, Button0, blue green and red gpios are defined, but their pinmuxes are not set up in board.c 

As discussed earlier today, we plan to put full support for all peripherals for this board so that M4 app (Zephyr) can be run directly without the bootloader, hence:
- enabling all peripherals
- loading the FPGA code

JDuchniewicz, 2023-06-14:
Might be useful here https://www.zephyrproject.org/how-to-set-up-esp32-gpio-pins-in-zephyr-rtos/