Willmish/bibop-zephyr-eos-s3 #14: Use HAL for I2C communication

JDuchniewicz, 2023-06-10:
Just use HAL from antmicro to test that I2C communication works.
https://github.com/antmicro/hal_quicklogic

Willmish, 2023-06-13:
Just ;) .

So we got it working, through some pain and reading the manual (https://www.quicklogic.com/wp-content/uploads/2020/06/QL-S3-Technical-Reference-Manual.pdf) page 339, section 36.1 .

We are enabling Clock 01  and Clock 08 X1, do we need X04?

Willmish, 2023-06-14:
Got HAL from qorc-sdk working on qorc apps, and on zephyr apps too, however on qorc-sdk adding a wait for FFE to pick up data when transmitting to I2C device was necessary, see: https://github.com/JDuchniewicz/qorc-sdk/commit/489b5842096d79a0e1bf209a9f98da8fb8fb37a1

However this for some reason is not necessary for Zephyr and works without it? Adding it just in case, cause looks like a race condition which is just more likely on FreeRTOS (maybe less things are happening, so less threads to interleave and more likely the same thread runs very fast and FFE/CRU has no time to clean those regs?)

JDuchniewicz, 2023-06-14:
Yes, this definitely needs some further debugging, let's leave it on Zephyr in any case.

Willmish, 2023-06-22:
Also; this needs the currently fixed clock dividor to be read, rather than assumed t obe 6: https://github.com/Willmish/hal_quicklogic/blob/hal_i2c/HAL/src/eoss3_hal_i2c.c#L147C8-L147C8